espressif/esp32-arduino-lib-builder #239: Can't build the library

jsmith173, 2024-11-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

No build errors

### Actual behavior (suspected bug)

Build errors: missing esp_insights

### Error logs or terminal output

```shell
CMake Warning at esp-idf/tools/cmake/project.cmake:500 (message):
  enabled ccache in build but ccache program not found
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at esp-idf/tools/cmake/build.cmake:548 (message):
  ERROR: Because espressif/esp_matter (1.3.0) depends on
  espressif/esp_insights (1.0.1)

   and no versions of espressif/esp_matter match >1.3.0,<2.0.0, espressif/esp_matter (>=1.3.0,<2.0.0) requires espressif/esp_insights (1.0.1).

  So, because project depends on both espressif/esp_matter (^1.3.0) and
  espressif/esp_insights (^1.2.1), version solving failed.

Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:6 (project)
```

### Steps to reproduce the behavior

./build.sh -t esp32,esp32c3 -D none

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22

### Shell

Bash

me-no-dev, 2024-11-01:
seems you are using wrong branches for the lib-builder and Arduino. Use the `release.v5.3` branch for the latest code and `release/v5.1` for Arduino v3.0.x compatibility

jsmith173, 2024-11-01:
Should I have to install esp-matter?

me-no-dev, 2024-11-01:
if you use the proper branch, it will install everything on it's own (`release.v5.3`)

jsmith173, 2024-11-02:
Thanks. It works basically.

Now I'd like to compile the library with debug info
In defconfig.common I have added the following line
> CONFIG_COMPILER_OPTIMIZATION_DEBUG=y

(and removed the line with optimize for size)

Will this compile with debug info? I'm not sure
I disassembled the arduino compiled elf but it doesn't contain code lines (for example for esp_timer_init which is not in ROM).
(Arduino compile now with -Og)

rvbc1, 2024-11-04:
> seems you are using wrong branches for the lib-builder and Arduino. Use the `release.v5.3` branch for the latest code and `release/v5.1` for Arduino v3.0.x compatibility

currently branch `release/v5.1`, cannot be compiled because of error:

```
In file included from /home/rvbc-/storage/data/Git/ST-Git/esp32-arduino-lib-builder/components/arduino/libraries/Zigbee/src/ZigbeeEP.h:5,
                 from /home/rvbc-/storage/data/Git/ST-Git/esp32-arduino-lib-builder/components/arduino/libraries/Zigbee/src/ep/ZigbeeThermostat.h:8,
                 from /home/rvbc-/storage/data/Git/ST-Git/esp32-arduino-lib-builder/components/arduino/libraries/Zigbee/src/ep/ZigbeeThermostat.cpp:1:
/home/rvbc-/storage/data/Git/ST-Git/esp32-arduino-lib-builder/components/arduino/libraries/Zigbee/src/ZigbeeCore.h:8:10: fatal error: esp_zigbee_core.h: No such file or directory
    8 | #include "esp_zigbee_core.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
[2417/2424] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/firmware_msc_fat.c.obj
[2418/2424] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/chip-debug-report.cpp.obj
ninja: build stopped: subcommand failed.
Missing "esp_zigbee_core.h" file name found in the following component(s): espressif__esp-zigbee-lib(/home/rvbc-/storage/data/Git/ST-Git/esp32-arduino-lib-builder/managed_components/espressif__esp-zigbee-lib/include/esp_zigbee_core.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.
```

me-no-dev, 2024-11-05:
please try again

jsmith173, 2024-11-05:
Thanks, my problem solved