espressif/esp32-arduino-lib-builder #241: Issues building libraries

HamzaHajeir, 2024-11-05:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hi, I've branched out from `release/v5.1` directly and edited to build on GitHub actions (branch named `test-release/v5.1`, however the build failed.

- Full log:  https://github.com/HamzaHajeir/esp32-arduino-lib-builder/actions/runs/11669768171/job/32492476390
- IDF branch used: release/v5.1
- Arduino branch used: idf-release/v5.1
- Error:
> /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Zigbee/src/ZigbeeCore.h:8:10: fatal error: esp_zigbee_core.h: No such file or directory
>     8 | #include "esp_zigbee_core.h"
>       |          ^~~~~~~~~~~~~~~~~~~
> compilation terminated.
> ninja: build stopped: subcommand failed.
> Missing "esp_zigbee_core.h" file name found in the following component(s): espressif__esp-zigbee-lib(/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/managed_components/espressif__esp-zigbee-lib/include/esp_zigbee_core.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.

It seems that the Zigbee Library was [added recently to the Arduino core](https://github.com/espressif/arduino-esp32/pull/10265).

### Actual behavior (suspected bug)

.

### Steps to reproduce the behavior

.

### Project release version

.

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

.

### Shell

Bash

me-no-dev, 2024-11-05:
issue should already be fixed. Try again please

closed 2024-11-05 by me-no-dev

HamzaHajeir, 2024-11-05:
> issue should already be fixed. Try again please

Correct, I had my tests last night. However, the error now gets in compiling zb libraries:

Github Run: https://github.com/HamzaHajeir/esp32-arduino-lib-builder/actions/runs/11682657369/job/32530238858 

** Might it be some bug in the sources?

```
[2417/2424] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/Zigbee/src/ZigbeeHandlers.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/Zigbee/src/ZigbeeHandlers.cpp.obj 
ccache /Users/runner/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -DESP32_ARDUINO_LIB_BUILDER -DESP_PLATFORM -DIDF_VER=\"v5.1.4-990-gf12cfcbcdb-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DOPENTHREAD_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DOPENTHREAD_FTD=1 -DOPENTHREAD_PROJECT_LIB_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/build/config -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/variants/esp32h2 -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/cores/esp32 -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/ArduinoOTA/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/AsyncUDP/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/BLE/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/BluetoothSerial/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/DNSServer/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/EEPROM/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/ESP_I2S/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/ESP_NOW/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/ESP_SR/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/ESPmDNS/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Ethernet/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/FFat/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/FS/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/HTTPClient/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/HTTPUpdate/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Insights/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/LittleFS/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/NetBIOS/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Network/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/OpenThread/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/PPP/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Preferences/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/SD_MMC/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/SD/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/SimpleBLE/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/SPIFFS/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/SPI/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Ticker/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Update/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/USB/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/WebServer/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/NetworkClientSecure/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/WiFiProv/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/libraries/Wire/src -I/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/compo [... cut ...]
   29 | static esp_err_t zb_attribute_set_handler(const esp_zb_zcl_set_attr_value_message_t *message) {
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
[2418/2424] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/sketch.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/build/log/idf_py_stderr_output_6266 and /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/build/log/idf_py_stdout_output_6266
```

me-no-dev, 2024-11-05:
@P-R-O-C-H-Y  is still working on it