espressif/esp32-arduino-lib-builder #243: ESP32S3 Compilation fails

weoiss1998, 2024-11-11:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

compile successfully

### Actual behavior (suspected bug)

fails

### Steps to reproduce the behavior

Use docker Image

### Project release version

release 5.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

weoiss1998, 2024-11-11:
[image]

me-no-dev, 2024-11-12:
are you using the `release/v5.1` branch of this repository, or are you using `master`? You should use `release/v5.1` for now

weoiss1998, 2024-11-12:
I use the Powershell Script of the Release/v5.1 branch

lucasssvaz, 2024-11-12:
@weoiss1998 Are you using any custom sdkconfig ?

weoiss1998, 2024-11-12:
I tried both, the original Version and the modified version from:
https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/performance/reduce-boot-time.html

weoiss1998, 2024-11-13:
I tried it today and after some misses it's working again. I have attached the error logs before it starts working again.
[idf_py_stderr_output_1019.zip](https://github.com/user-attachments/files/17738760/idf_py_stderr_output_1019.zip)

The solution for me is to try again until it compiles successfully. I didn't change any settings, I just clicked on ‘Compile Static Libraries’ again and again. Is it possible to reuse the container from the Powershell script and if so, how?