espressif/esp32-arduino-lib-builder #244: Issue building libraries with 3.1-RC2

blorp8000, 2024-11-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Tried building with
`./build.sh -t esp32c3 -A release/v3.1.x -I release/v5.3 -c ~/Documents/Arduino/hardware/espressif/esp32`

### Actual behavior (suspected bug)

Building failed with `FAILED: esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/esp_mmu_map.c.obj ` and `ninja: build stopped: subcommand failed.
ninja failed with exit code 1`.

idf_py_stdout_output_60054 log output is attached: (i edited some lines out to stay in the character limit)

### Error logs or terminal output

```shell
[1/2213] Generating project_elf_src_esp32c3.c
[2/2213] Generating /Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/build/esp-idf/esp_system/ld/memory.ld linker script...
[3/2213] Generating /Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/build/esp-idf/esp_system/ld/sections.ld.in linker script...
[4/2213] Building C object esp-idf/espcoredump/CMakeFiles/__idf_espcoredump.dir/src/core_dump_init.c.obj
[5/2213] Building C object esp-idf/espcoredump/CMakeFiles/__idf_espcoredump.dir/src/core_dump_common.c.obj
[6/2213] Building C object esp-idf/espcoredump/CMakeFiles/__idf_espcoredump.dir/src/core_dump_binary.c.obj

[...]

[577/2213] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_os_func_app.c.obj
[578/2213] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/port/esp32c3/ext_mem_layout.c.obj
[579/2213] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/esp_cache.c.obj
[580/2213] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/heap_align_hw.c.obj
[581/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_common.c.obj
[582/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_clock_init.c.obj
[583/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_mem.c.obj
[584/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_common_loader.c.obj
[585/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_random.c.obj
[586/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_efuse.c.obj
[587/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/flash_encrypt.c.obj
[588/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/secure_boot.c.obj
[589/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_random_esp32c3.c.obj
[590/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/flash_qio_mode.c.obj
[591/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash.c.obj
[592/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash_config_esp32c3.c.obj
[593/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_utility.c.obj
[594/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/flash_partitions.c.obj
[595/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp_image_format.c.obj
[596/2213] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32c3/esp_efuse_table.c.obj
[597/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/idf/bootloader_sha.c.obj
[598/2213] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32c3/secure_boot_secure_features.c.obj
[599/2213] Linking C static library esp-idf/esp_wifi/libesp_wifi.a
[600/2213] Linking C static library esp-idf/esp_coex/libesp_coex.a
[601/2213] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/esp_mmu_map.c.obj
FAILED: esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/esp_mmu_map.c.obj 
ccache /Users/niki/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.3.1-638-ga0f798cfc4-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/build/config -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_mm/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/newlib/platform_include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/config/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/config/include/freertos -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/config/riscv/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/freertos/esp_additions/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/dma/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/ldo/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32c3/. -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32c3/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/heap/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/log/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/soc/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/soc/esp32c3 -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/soc/esp32c3/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/hal/platform_port/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/hal/esp32c3/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/hal/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_rom/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_rom/include/esp32c3 -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_rom/esp32c3 -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_common/include -I/Users/niki/Documents/Arduino/hardware/espressif/esp32-arduino-lib-builder/esp-idf/components/esp_system/include -I/Users/niki/Documents/Arduino/hardware/espressi [... cut ...]
[... 123 lines cut ...]
### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Sonoma

### Shell

ZSH

### Additional context

Tried this process of building custom libs, because:
1. arduino-esp32-RC2 for Arduino-IDE does not have CONFIG_LWIP_PPP_SERVER_SUPPORT enabled
2. installing arduino-esp32-RC2 as a component in ESP-IDF v5.3.1 doesn't work since component requirements are still set at <v5.1 and v5.2>

me-no-dev, 2024-11-13:
you need to use the `release/v5.3` branch of this repository and not specify `-A` and `-I`

closed 2024-11-13 by blorp8000

blorp8000, 2024-11-13:
That worked!

Fwiw, I also had to make sure that "Finder" in Mac OS was not calculating any folder sizes of esp32-arduino-lib-builder.
I previously deleted all .DS_Store files in the directory and all subfolders with
`find . -name ".DS_Store" -print -delete`
but during the build process they kept popping up again, which resulted in a CMake error.
Disabling "calculate all sizes" in Finder fixed that.