espressif/esp32-arduino-lib-builder #247: Fail when process /tools/install-arduino.sh as bug in config.sh

twischi, 2024-11-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Clone all needed stuff from GH resources before start building 

### Actual behavior (suspected bug)

install-arduiono.sh fails, as $AR_REPO_URL is invalid url.

* Installing/Updating ESP-IDF and all components...
Updating TinyUSB...
Already up to date.
Updating ESP32 Arduino...
AR_BRANCH='release/v3.1.x'
error: pathspec 'release/v3.1.x' did not match any file(s) known to git

### Error logs or terminal output

```shell
* Installing/Updating ESP-IDF and all components...
Updating TinyUSB...
Already up to date.
Updating ESP32 Arduino...
Current Branch: release/v5.3
AR_BRANCH='release/v5.3'
Already on 'release/v5.3'
Your branch is up to date with 'origin/release/v5.3'.
Already up to date.
ESP-IDF is not installed! Installing local copy
Cloning into 'esp-idf'...
Username for 'https://github.com': ^[[A^[[A^C
thomas@iMac2TW esp32-arduino-lib-builder % git checkout master                                                              
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
thomas@iMac2TW esp32-arduino-lib-builder % ./build.sh -t esp32h2,esp32s3,esp32c3,esp32 -A "release/v3.1.x" -I "release/v5.3"                   
TARGET(s): esp32h2 esp32s3 esp32c3 esp32
* Installing/Updating ESP-IDF and all components...
Updating TinyUSB...
Already up to date.
Updating ESP32 Arduino...
AR_BRANCH='release/v3.1.x'
error: pathspec 'release/v3.1.x' did not match any file(s) known to git
thomas@iMac2TW esp32-arduino-lib-builder % git checkout release/v5.3                                                        
Switched to branch 'release/v5.3'
Your branch is up to date with 'origin/release/v5.3'.
thomas@iMac2TW esp32-arduino-lib-builder % ./build.sh -t esp32h2,esp32s3,esp32c3,esp32 -A "release/v3.1.x" -I "release/v5.3"
TARGET(s): esp32h2 esp32s3 esp32c3 esp32
* Installing/Updating ESP-IDF and all components...
Updating TinyUSB...
Already up to date.
Updating ESP32 Arduino...
AR_BRANCH='release/v3.1.x'
error: pathspec 'release/v3.1.x' did not match any file(s) known to git
```

### Steps to reproduce the behavior

1. Cloned this repo (as normal user)
2. Launch ./build.sh
   (Both fails no arguments & "-t esp32h2,esp32s3,esp32c3,esp32 -A "release/v3.1.x" -I "release/v5.3"")
3. Abort when try to load with /tools/install-arduino.sh

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

Monterey

### Shell

Bash

### Additional context

I found and fixed the bug already!

the problem is in:
https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/config.sh

The variable   $GITHUB_REPOSITORY_OWNER    is not set, so are AR-variables are invalid. 

# Owner of the target ESP32 Arduino repository
AR_USER="$GITHUB_REPOSITORY_OWNER"

# The full name of the repository
AR_REPO="$AR_USER/arduino-esp32"
IDF_REPO="$AR_USER/esp-idf"
AR_LIBS_REPO="$AR_USER/esp32-arduino-lib-builder"

---- If fixed this with just change to:
# Owner of the target ESP32 Arduino repository
AR_USER="espressif"

me-no-dev, 2024-11-25:
The branch 'release/v3.1.x' does exists in Arduino just fine (go have a look :)). Are you using the 'release/v5.3' branch of the lib-builder?

twischi, 2024-11-25:
You are right for branch: 'release/v5.3' 

I refer more to the „master,“ where the variable for AR_USER is still used. But GITHUB_REPOSITORY_OWNER is not defined in the complete Repo > guess > defined in bash.

https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/config.sh <https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/config.sh>

Greets
Thomas

PR #253 (2024-12-10): fix(config): Add fallback value to AR_USER
## Description

Add missing fallback value to `AR_USER`. This is required when running locally.

## Related

Closes #247

## Testing

Locally
closed 2024-12-10 by me-no-dev