espressif/esp32-arduino-lib-builder #288: Help: Known working version set for build

sourcesimian, 2025-03-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I would like to build Arduino core (esp32 only) from source so that I can minimise resource usage. I additionally need to minimise my downloads. I am attempting to build using the Docker container, with the espressif/esp-idf and espressif/arduino-esp32 repos volume mounted in from shallow clones that I perform before I run the build. And I am using LIBBUILDER_CLONE_SHALLOW=1, and LIBBUILDER_CLONE_BRANCH_OR_TAG. 

I can not seem to figure out a working combination of branch or tag versions for espressif/esp-idf, espressif/arduino-esp32 and espressif/esp32-arduino-lib-builder where the build completes successfully. I would preferably like to use a build where matter is available.

Please could you provide such a version set (also should I use branches or tags), and additionally tell me how I can figure this out for myself in the future.

Many thanks.

me-no-dev, 2025-03-16:
ESP-IDF `release/v5.4` + Arduino `master` + Lib-builder `master` will result in the current Arduino `3.2.0-RC2` build

sourcesimian, 2025-03-16:
Thank you. Using those versions and also others I seem to get the following error. Are you able to provide some guidance please?

```
CMake Error at esp-idf/tools/cmake/component.cmake:228 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 106

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /opt/esp/lib-builder/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Warning (dev) at
  /opt/esp/lib-builder/components/arduino/CMakeLists.txt:12 (if):

    if given arguments:

  

      "5.4.0" "AND" "NOT" ""

  

    An argument named "5.4.0" appears in a conditional statement.  Policy
    CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
    "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
    command to set the policy and suppress this warning.

  Call Stack (most recent call first):

    /opt/esp/lib-builder/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    /opt/esp/lib-builder/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:160 (__component_get_requirements)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  /opt/esp/lib-builder/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:52
  (__component_get_property):

    __component_get_property Function invoked with incorrect arguments for
    function named: __component_get_property

  Call Stack (most recent call first):

    /opt/esp/lib-builder/managed_components/espressif__network_provisioning/CMakeLists.txt:23 (idf_component_get_property)
    /opt/esp/lib-builder/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    /opt/esp/lib-builder/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:160 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:644 (__component_get_requirements)
  esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)
```

me-no-dev, 2025-03-17:
are your cmake and ninja up to date? Very weird error

sourcesimian, 2025-03-18:
Thank you for your rapid and astute assistance @me-no-dev.
I was trying to clone esp-idf as shallow and sparse, but I clearly missed some required sources.

closed 2025-03-18 by sourcesimian