espressif/esp32-arduino-lib-builder #292: Build error: project depends on both idf (4.4.*) and idf (>=5.0)

mamk95, 2025-03-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Expect the `build.sh` command to complete without errors, creating a build of the ESP32 libraries for Arduino similar to the official v2.0.17 libraries.

### Actual behavior (suspected bug)

Once it gets to `idf.py` it crashes with error "Because project depends on both idf (4.4.*) and idf (>=5.0), version solving failed".

### Error logs or terminal output

```shell
$ ./build.sh -t esp32 -A release/v2.x -I release/v4.4
...
Using Python interpreter in /home/MyUsername/.espressif/python_env/idf4.4_py3.11_env/bin/python
Checking if Python packages are up to date...
Python requirements from /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/esp-idf/components/esptool_py/esptool
  /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/esp-idf/components/espcoredump
  /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/esp-idf/components/partition_table
  /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/esp-idf/components/app_update
  /home/MyUsername/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /home/MyUsername/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /home/MyUsername/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
  /home/MyUsername/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin
  /home/MyUsername/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin
  /home/MyUsername/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin
  /home/MyUsername/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /home/MyUsername/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin
  /home/MyUsername/.espressif/python_env/idf4.4_py3.11_env/bin
  /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

* Skipping Target: esp32s3
* Skipping Target: esp32s2
* Skipping Target: esp32c3
* Target: esp32
* Build IDF-Libs: configs/defconfig.common;configs/defconfig.esp32;configs/defconfig.qio_ram;configs/defconfig.qio;configs/defconfig.80m
Executing action: idf_libs
Running cmake in directory /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DSDKCONFIG_DEFAULTS=configs/defconfig.common;configs/defconfig.esp32;configs/defconfig.qio_ram;configs/defconfig.qio;configs/defconfig.80m -DCCACHE_ENABLE=0 /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder"...
-- Found Git: /usr/bin/git (found version "2.39.5")
-- Component directory /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp-insights does not contain a CMakeLists.txt file. No component will be added
-- Component directory /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/MyUsername/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/MyUsername/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/MyUsername/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
NOTICE: Skipping optional dependency: espressif/mdns
........HINT: Please check manifest file of the following component(s): esp-dl
CMake Error at esp-idf/tools/cmake/build.cmake:463 (message):
  WARNING: Component "idf" not found

  WARNING: Component "idf" not found

  ERROR: Because project depends on both idf (4.4.*) and idf (>=5.0), version
  solving failed.

Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!
See also "/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

### Steps to reproduce the behavior

```powershell
wsl --install Debian
wsl -d Debian
```

```bash
sudo apt update
sudo apt upgrade
sudo apt install git wget curl libssl-dev libncurses-dev flex bison gperf python3 python3-pip python3-setuptools python3-serial python3-click python3-cryptography python3-future python3-pyparsing python3-pyelftools python3-venv python-is-python3 cmake ninja-build ccache jq libusb-1.0-0
git clone -b release/v4.4 https://github.com/espressif/esp32-arduino-lib-builder
cd esp32-arduino-lib-builder
python3 -m pip config set global.break-system-packages true
./build.sh -t esp32 -A release/v2.x -I release/v4.4
```

### Project release version

esp32-arduino-lib-builder v4.4, Arduino-ESP32 v2.x, ESP-IDF v4.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10 running Debian in WSL. Also tried Ubuntu in WSL with same result.

### Shell

Bash

### Additional context

I'm trying to create the ESP32 libraries to replace the official v2.0.17 libraries, with the exception/improvement of supporting more than 3 bluetooth connections. I cannot use the new v3 libraries due to LilyGo T-Watch library incompatibility - only v2 libraries are supported. I've tried the above steps multiple times with clean Debian and Ubuntu in WSL (Windows Subsystem for Linux).

If it's of any interest, I also get the same error when running `idf.py build` or `idf.py menuconfig`:
```bash
. ./tools/install-esp-idf.sh -t esp32 -A release/v2.x -I release/v4.4
idf.py build
```

Any help would be greatly appreciated!

me-no-dev, 2025-03-31:
master branch creates libs only for the current arduino version (3.2.0 in our case). For compiling other versions, you need to use the relevant brach of this repository. In your case that would be `release/v4.4`. So just checkout that branch and run `./build.sh -t esp32` and you will get 2.0.17 compatible libs

closed 2025-03-31 by me-no-dev

mamk95, 2025-03-31:
I'm already using the v4.4 branch, see my git checkout command in the original comment above: `git clone -b release/v4.4 https://github.com/espressif/esp32-arduino-lib-builder`

It might seem like the problem is some of the other components requiring idf >=5.0, like esp-dl, esp-rainmaker, esp_littlefs and freemodbus:
```
$grep -r ">=5.0" --include="idf_component.yml" .
./components/esp-rainmaker/components/esp_rainmaker/idf_component.yml:      - if: "idf_version >=5.0"
./components/esp_littlefs/idf_component.yml:  idf: ">=5.0"
./esp-idf/components/freemodbus/test/tcp/mb_tcp_master/main/idf_component.yml:      - if: "idf_version >=5.0"
./esp-idf/components/freemodbus/test/tcp/mb_tcp_slave/main/idf_component.yml:      - if: "idf_version >=5.0"
```

I'm getting a bit further doing this:
```
cd components/esp-dl
git checkout idfv4.4
cd ../../components/esp-rainmaker
git checkout idf_4_x_compat
git submodule update --recursive
cd ../../components/esp_littlefs
git checkout -b branch-based-on-v1.16.4 v1.16.4
cd ../..
```

But I'm still getting errors:
```
$ ./build.sh -t esp32 -A release/v2.x -I release/v4.4
* Installing/Updating ESP-IDF and all components...
Updating ESP32 Arduino...
Already on 'release/v2.x'
Your branch is up to date with 'origin/release/v2.x'.
Already up to date.
Updating ESP32 Camera...
Already up to date.
Updating ESP-DL...
Updating ESP-LITTLEFS...
There is no tracking information for the current branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

If you wish to set tracking information for this branch you can do so with:

    git branch --set-upstream-to=origin/<branch> branch-based-on-v1.16.4
```

me-no-dev, 2025-03-31:
You can try to set the components version to what is here: https://github.com/espressif/arduino-esp32/blob/release/v2.x/tools/sdk/versions.txt
Preferably do that in `tools/update-components.sh` to ensure that `./build.sh` without `-s` will succeed

mamk95, 2025-03-31:
The build continues but later fails with the following error:

Short version:
```
In file included from /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMaker.h:20,
                 from /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMakerUtils.h:17,
                 from /mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMakerUtils.cpp:1:
/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src/RMakerQR.h:19:10: fatal error: qrcode.h: No such file or directory
 #include <qrcode.h>
          ^~~~~~~~~~
compilation terminated.
```

<details>

<summary>Full error log</summary>

```
[1607/1851] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMaker.cpp.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/RainMaker/src/RMaker.cpp.obj
/home/MyUsername/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/build/config -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/variants/esp32 -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/cores/esp32 -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/ArduinoOTA/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/AsyncUDP/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/BLE/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/BluetoothSerial/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/DNSServer/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/EEPROM/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/ESP32/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/ESPmDNS/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/Ethernet/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/FFat/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/FS/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/HTTPClient/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/HTTPUpdate/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/LittleFS/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/Insights/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/I2S/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/NetBIOS/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/Preferences/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/RainMaker/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/SD_MMC/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/SD/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/SimpleBLE/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/SPIFFS/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/SPI/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/Ticker/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/Update/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/USB/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/WebServer/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/WiFiClientSecure/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/WiFiProv/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/libraries/Wire/src -I/mnt/c/Users/MyUsername/Documents/ESP32LibBuilder/esp32-arduino-lib-builder/components/arduino/cores/esp32/libb64 -I/mnt/c/User [... cut ...]
 fi
 if [ $? -ne 0 ]; then exit 1; fi
(END)
```

Thank you for your help @me-no-dev - I truly appreciate it!

This problem happened on a clean Debian install using the official instructions. As far as I can see, there were no user errors in the installation. Based on that, it seems likely that users will start having this problem more and more from now on, as dependencies start to remove support for IDF 4. You might consider patching the `update-components.sh` file to help avoid future problems.

me-no-dev, 2025-03-31:
`qrcode.h` comes from RainMaker (or a submodule). In the versions I pointed out, there are the commit hashes of the exact used ones for 2.0.17. It is possible that the qrcode code has been removed or moved to another repo.

We (Espressif ESP-IDF and Arduino) no longer support anything related to ESP-IDF 4.4. We only keep the branches for reference. We do accept pull requests for cases like yours, but do not otherwise spend time on them.

civet, 2025-10-31:
Unfortunately, v4.4 is no longer supported, and this script still has issues.
@mamk95 Thank you for providing help. By fixing the versions of components in `tools/update-components.sh`,  now v4.4 can finally be compiled normally.

Here is the script I have modified. You can fix the version of the components as needed by changing the variables.
[update-components.sh](https://github.com/user-attachments/files/23253920/update-components.sh)