espressif/esp32-arduino-lib-builder #294: Build fails on arduino_tinyusb

swiergot, 2025-04-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Build succeeds

### Actual behavior (suspected bug)

Build fails early on arduino_tinyusb.

### Error logs or terminal output

```shell
FAILED: esp-idf/arduino_tinyusb/CMakeFiles/__idf_arduino_tinyusb.dir/src/dcd_dwc2.c.obj 
ccache /home/swiergot/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DESP32_ARDUINO_LIB_BUILDER -DESP_PLATFORM -DIDF_VER=\"v5.4.1-361-g72de7111cc-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/swiergot/esp32-arduino-lib-builder/build/config -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos -I/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src -I/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/include -I/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/hw/bsp -I/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/device -I/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/portable/synopsys/dwc2 -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/newlib/platform_include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/config/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/config/include/freertos -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/config/riscv/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/freertos/esp_additions/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/esp32p4 -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/dma/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/ldo/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/debug_probe/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32p4/. -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32p4/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32p4/private_include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/heap/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/log/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/soc/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/soc/esp32p4 -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/soc/esp32p4/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/soc/esp32p4/register -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/hal/platform_port/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/hal/esp32p4/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/hal/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_rom/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_rom/esp32p4/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_rom/esp32p4/include/esp32p4 -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_rom/esp32p4 -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_common/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_system/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/include/riscv -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/include/private -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/riscv/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/esp_timer/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/include/apps -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/include/apps/sntp -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/port/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/port/freertos/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32xx/include -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/swiergot/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32xx/include/sys -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Os -freorder-blocks -Wwrite-strings -fstack-protector -fmacro-prefix-map=/home/swiergot/esp32-arduino-lib-builder=. -fmacro-prefix-map=/home/swiergot/esp32-arduino-lib-builder/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -DCFG_TUSB_MCU=OPT_MCU_ESP32P4 -DCFG_TUSB_DEBUG=0 -Wno-type-limits -MD -MT esp-idf/arduino_tinyusb/CMakeFiles/__idf_arduino_tinyusb.dir/src/dcd_dwc2.c.obj -MF esp-idf/arduino_tinyusb/CMakeFiles/__idf_arduino_tinyusb.dir/src/dcd_dwc2.c.obj.d -o esp-idf/arduino_tinyusb/CMakeFiles/__idf_arduino_tinyusb.dir/src/dcd_dwc2.c.obj -c /home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c: In function 'dma_device_enabled':
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c:105:43: error: 'dwc2_regs_t' has no member named 'ghwcfg2_bm'; did you mean 'ghwcfg4_bm'?
  105 |   return CFG_TUD_DWC2_DMA_ENABLE && dwc2->ghwcfg2_bm.arch == GHWCFG2_ARCH_INTERNAL_DMA;
      |                                           ^~~~~~~~~~
      |                                           ghwcfg4_bm
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c: In function 'edpt_schedule_packets':
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c:388:37: error: 'dwc2_regs_t' has no member named 'dsts_bm'; did you mean 'dsts'?
  388 |     const uint32_t odd_now = (dwc2->dsts_bm.frame_number & 1u);
      |                                     ^~~~~~~
      |                                     dsts
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c: In function 'dcd_init':
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c:435:15: error: 'dwc2_regs_t' has no member named 'ghwcfg2_bm'; did you mean 'ghwcfg4_bm'?
  435 |     if (dwc2->ghwcfg2_bm.hs_phy_type == GHWCFG2_HSPHY_ULPI) {
      |               ^~~~~~~~~~
      |               ghwcfg4_bm
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c: In function 'handle_bus_reset':
/home/swiergot/esp32-arduino-lib-builder/components/arduino_tinyusb/src/dcd_dwc2.c:47:43: error: 'dwc2_regs_t' has no member named 'ghwcfg2_bm'; did you mean 'ghwcfg4_bm'?
   47 |   #define DWC2_EP_COUNT(_dwc2)  ((_dwc2)->ghwcfg2_bm.num_dev_ep + 1)
[... cut ...]
In general I'm trying to reduce my image size so I thought I would rebuild arduino-esp32 without a few unneeded things. And that's also not going well. For instance, I have commented out `CONFIG_LWIP_PPP_*` options in `configs/defconfig.common` but they are still in `sdkconfig`. How to get rid of that?

Another matter is how to build exactly version 3.2.0 with esp-idf 5.4.1? There is branch for version 3.2.0. There isn't also a branch for that particular esp-idf version, just `release/v5.4` but that's probably work in progress for the next 5.4.x.

Any help greatly appreciated.

me-no-dev, 2025-04-16:
Updated master. Please pull the latest changes and try again :)

me-no-dev, 2025-04-16:
3.2.0 is built with ESP-IDF tag 5.4.1. That tag was released from the `release/v5.4` branch also.

swiergot, 2025-04-17:
Hi, I can confirm that TinyUSB is fixed. I have managed to build version 3.2.0 with `./build.sh -A 3.2.0 -I v5.4.1` but build scripts need to be changed slightly. I will submit a PR. Thanks again for help.

closed 2025-04-17 by swiergot